fritzing/fritzing-parts #403: The sparkfun-relay packet relay-g5v, variant g5q appears as a red rectangle in breadboard view.

dac1e, 2024-12-18:
## Current Behaviour
The sparkfun-relay appears a as a red rectangle in breadboard view and cannot be relocated with the mouse or cursor keys.
See the following screenshot:
[image]
See zipped example fzz file.
[sparkfun-relay.zip](https://github.com/user-attachments/files/18183761/sparkfun-relay.zip)

**Build:**
Version 1.0.4 (CD-2088-0-a8c6ef7c 2024-10-07) 64 [Qt 6.5.3]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Drop Family 'sparfun relay (spdt), packet relay-g5v, variant g5q' onto  breadboard view
- See how  relay is rendered.

## Expected Behaviour
- Relay should be rendered correctly, not as red rectangle.

KjellMorgenstern, 2024-12-18:
I guess it was intended to look like this:

[image]

dac1e, 2024-12-19:
Yes, that's correct. Thank you.

RAPTOR7762, 2025-02-22:
https://github.com/fritzing/fritzing-parts/issues/406 
Please also look into this issue 🙂