fritzing/fritzing-parts #409: fc-51 parts in core parts have a problem in pcb.

vanepp, 2025-03-20:
The two fc-51 parts (simulation off and simulation on) have pcb pins reversed. See forum post

https://forum.fritzing.org/t/parts-are-mirrored-in-pcb/27736

Here are two parts that correct the problem, but I expect there is going to be an issue with obsoleting the existing parts (although hopefully I am wrong!) Since the pcb pin position has changed without the pin number changing I think the replacement part will route incorrectly in an existing sketch and I don't see a way around it, as the connector Id isn't changing in the fzp file. Hopefully you do see a way around it. I also took the opportunity to update the parts to meet the graphics standards. The two parts are here (as well as in the forum post.) Remove the trailing .zip to get the .fzpz file. 

[fc-51-IR-Sensor.fzpz.zip](https://github.com/user-attachments/files/19376845/fc-51-IR-Sensor.fzpz.zip)

[fc-51-IR-Sensor_On.fzpz.zip](https://github.com/user-attachments/files/19376859/fc-51-IR-Sensor_On.fzpz.zip)

KjellMorgenstern, 2025-03-21:
We could
- add the new fc-51 sensor parts
- move reverse orientated fc-51 to obsolete
- *not* set a replasedby id on the obsolete part, so existing circuits will just keep it. We assume these circuits already have been validated with the existing VCC - GND - OUT order
- unlist the old and list the new parts in the core and the sim bins.

vanepp, 2025-03-21:
That sounds like a good solution if you are agreeable to it. The issue I see is that parts that are routed in pcb in a sketch will suddenly break as the routing will change (pins 3 2 1 will become 1 2 3 and the traces will likely cross.) Unfortunately I don't see a way around that though. Ah! I think I see, the old sketch will still work because the old part is in obsolete and will be used (but be in the wrong order but still work as it did) and because it isn't obsoleted you won't be prompted to update it.  Yes that seems to be a very good solution that won't break anything unexpected as far as I can see. Good thinking (I was hoping you would be able to think of something I hadn't thought of and you did!)

RAPTOR7762, 2025-03-30:
Yes, I agree with Peter's suggestion.

@KjellMorgenstern shall we go ahead with obsoleting the part (or will cause anything catastrophic?)

RAPTOR7762, 2025-04-01:
So like this (right-click --> save image as... to download the SVG)

[image]

KjellMorgenstern, 2025-04-02:
@RAPTOR7762 It should be a new part, so it should also not share the SVG with the old part.

What I ussually do:
1. get a (partial) uuid.
```
$ uuid | cut -f1 -d-
fc2fdbec

```

2. Append the id to the new ModuleID , to the fzp filename, and to the svg filesnames
3. Copy the new files to the core and svg folders
4. git add the files
5. Enlist the new part in a matching group and bin in the bin folder.
6. Update the Fritzing version of the part.
7. Update the part version of the part
8. commit the first version

Then test the part, fix any issues

RAPTOR7762, 2025-04-02:
>Append the id to the new ModuleID

Just to confirm -- do I add it to the front like this

[image]

>Update the part version of the part
>Update the Fritzing version of the part.

As this is my first time contributing a fix to this rrepo, what do you mean by that? I do have the latest part loaded in the "MINE" bin.

As well before I make a catastrophic mistake, what is going to happen to the existing part? Will it get obsoleted, or will the part get deleted entirely?

KjellMorgenstern, 2025-04-02:
Any change will only become effective once approved and released. Until then, myself and probably others will review the part. There is near zero chance for a catastrophic change, even if you would intentionally try craft one.
 
Basically you need to mimic the folder structure of other parts in this repo, and copy the files there. If you have never worked with git before, I recommend going through some tutorials.

RAPTOR7762, 2025-04-02:
OK, I'll try

RAPTOR7762, 2025-04-03:
Weird, I had uploaded all the relevant files. Am I missing something? 

@KjellMorgenstern can you help me add the files to this REPO?

## fc-f1_ON

**Part File**
[part.fc-51-IR-Sensor_On_3.fzp.txt](https://github.com/user-attachments/files/19587483/part.fc-51-IR-Sensor_On_3.fzp)

**Breadboard SVG**
[image]

**Icon SVG**
[image]

**PCB SVG**
[image]

**Schematic SVG**
[image]

## fc-f1

**Part File**
[part.fc-51-IR-Sensor_On_3.fzp.txt](https://github.com/user-attachments/files/19587522/part.fc-51-IR-Sensor_On_3.fzp.txt)

**Breadboard SVG**
[image]

**Icon SVG**
[image]

**PCB SVG**
[image]

**Schematic SVG**
[image]