fritzing/fritzing-parts #416: SparkFun 7 segment display has issues

RAPTOR7762, 2025-04-28:
This is version 1.0.5

[image]

This is version 0.9.10

[image]

As well in schematic the wire terminates to the centre of the pin

RAPTOR7762, 2025-04-28:
@KjellMorgenstern any way to contribute this
[7-SEGMENT-DISPLAY-fixed.fzpz](https://forum.fritzing.org/uploads/short-url/iqS0h7lRCJ8ASXjztGdG6JuBlle.fzpz) (7.4 KB)

KjellMorgenstern, 2025-05-22:
The module id of the broken part is "sparkfun-displays-7-segment-display-pth.fzp"

When looking into the PCB svg, there is this structure:

```
 <g gorn="0.1" id="copper1" >
        <circle connectorname="1" cx="1.2216" cy="16.2216" fill="none" gorn="0.1.0" id="connector0pad" r="0.6324" stroke="#F7BD13" stroke-width="0.6148" />
....
        <g gorn="0.1.10" id="copper0" >
            <circle connectorname="1" cx="1.2216" cy="16.2216" fill="none" gorn="0.1.10.0" id="connector0pad" r="0.6324" stroke="#F7BD13" stroke-width="0.6148" />
...
        </g>
  </g>
```

This seems broken: The element with the ID "connector0pad" exists two times.
On time in the group with id copper1, and a second time in the group copper0.
In addition the cooper0 group is subgroup of the copper1 group.

This means there are these xpaths by ID:
```

copper1->connector0pad
copper1->copper0->connector0pad
```

However, when looking for the layers, Fritzing mathes the pattern more like

`<layer>/**/<connector>`

This means there are three possible matches:

**copper1**->**connector0pad**
**copper1**->copper0->**connector0pad**
copper1->**copper0**->**connector0pad**

The fixed version doesn't try to clean this up. Can you explain what the fixed version changed?

RAPTOR7762, 2025-06-04:
OK @KjellMorgenstern So I've done the PCB

[image]

As well I've fixed the "wrong-connector-order" issue in Breadboard